EdaphicStudio/SystemVerilog #1: Clocking drive with macro generates incorrect syntax error

rekendahl, 2018-06-28:
The following code generates an error at the ';' of line 13 where it shouldn't. 
The error message in the comment marked with `//^` on line 14.

**Notice**: Line 11 (with `#100`) works while the one with a macro definition fails. 

```SystemVerilog

`define TCQ 100

module issue;
    always @(posedge clk) begin

        if (a == 1'b1) begin
            diff_pntr         <= 0;
        end else begin
            if (wr_en)
                diff_pntr <= #100 (wr_pntr - adj_rd_pntr_wr + 2'h1);  // **WORKS!**
            else if (wr_en)
                diff_pntr <= #`TCQ (wr_pntr - adj_rd_pntr_wr);
                //^ <expression> expected, got ';'
        end
    end
endmodule
```

EdaphicStudio, 2018-06-28:
Thank you for filling the issue. We can reproduce the error on our end. Will try to resolve in next release.

EdaphicStudio, 2018-06-29:
Interesting test case. The logic for resolving macros blindly took any `(` as the start of macro arguments even when the define had no actual arguments. Which meant that in the following code everything inside the parenthesis was passed to the macro and then simply dropped resulting in bad code.

```systemverilog
diff_pntr <= #`TCQ (wr_pntr - adj_rd_pntr_wr);
```

The short-term solution, _which will be available in beta 4_, is to look up if a macro needs arguments and only then look for a `(`

The medium-term solution is to error-check that the number of arguments used by a macro matches the expected number from the define and flag errors as appropriate. Today the plug-in assumes you pass it the correct amount of arguments and doesn't flag errors for it.

closed 2018-06-29 by EdaphicStudio