EdaphicStudio/SystemVerilog #3: UVM Macro `uvm_component_utils` reports error

rekendahl, 2018-07-30:
The macro `uvm_components_utils` reports an error even though it seemingly is expanded correctly. I have attached a screenshot of the error.

This can be seen by using the UVM ubus example and opening the `test_lib.sv` file.

[image]

EdaphicStudio, 2018-07-30:
This bug was introduced with the fix for #2 

The fix didn't take into account defines from external sources when checking for the number of parameters. Working on a fix now.

EdaphicStudio, 2018-07-31:
Found cause - Simple bug where the check for arguments used a local list of defines from the file instead of the list of defines seen. Verified it works for all cases in regression. Will be part of next beta release.

closed 2018-07-31 by EdaphicStudio