EdaphicStudio/SystemVerilog #9: ctrl+alt+space strange behavior

VaginAY, 2018-09-23:
Good day!
Here is ctrl+alt+space in empty begin/end:
[image]
Here is ctrl+alt+space in begin/end with 'cnt':
[image]
If I put "cnt <= cnt - 1'b;" before insertion point, behavior = first case (no suggestions).

Best regards,
Alexander

EdaphicStudio, 2018-09-26:
Interesting find. Any chance you can share the example code as text? That way it's faster for me to create a regression test for it.

Off the top of my head, I think the issue is with being on a first and only line inside a block. The analyzer doesn't know if you are trying to create a variable or make a statement. It should auto-complete for both but options, but currently, it doesn't.  It's a little weird that it works in the second case. 

I will try to solve this test-case for the next release (aiming for early next week).

VaginAY, 2018-09-27:
[edaphicDebug.zip](https://github.com/EdaphicStudio/SystemVerilog/files/2422493/edaphicDebug.zip)

Test project for debug this issue.

EdaphicStudio, 2018-09-27:
Thank you for the example files I will make sure it works before release.

I'm making progress on this issue and have it working in my test area, but my fix breaks other things currently.

This issue forces a much needed but large optimization of the statement rule detector logic.  So clean-up from this will take me some more time. 

The core issue is that just typing an identifier doesn't match to any specific rule as it's matching many rules. None of these different rules have enough information to say "I'm it." When no rule matches then no code completion happens since code completion is content aware and only completes for objects in scope. This is not the intended functionality.

The solution is to restructure the statement detector to unify all rules that can start with an Identifier into a single master rule. This change has a lot of advantages beyond code completion, but the update cuts at the heart of the analyzer. 

So I'm not dealing with all the "corner case" syntaxes that I've found. The good thing is that with over 800 tests in regression I feel reasonably confident releasing this update once all tests work again.

Good bug!

EdaphicStudio, 2018-09-28:
This issue gets resolved in [BETA7](https://plugins.jetbrains.com/plugin/10695-systemverilog--edaphic-studio-llc/update/50428) just pushed to JetBrains' queue and should be available to users the week of October 1st.

Teaser gif:

[image]

closed 2018-09-28 by EdaphicStudio