EdaphicStudio/SystemVerilog #15: invalid formatting for interface class

edcote, 2018-10-15:
I am seeing incorrect formatting (parser reports syntax error) on the following:

~~~systemverilog
interface class #(type T = uvm_object) nvm_strategy_base;
  T m_strategy;
  pure virtual function void set_strategy(T t);
endclass
~~~

EdaphicStudio, 2018-10-15:
Nice find!

I have confirmed the error as well and looking at the rules we currently don't support interface classes. 

We will add support in the next release as this is an oversight on our part.

If you can please provide as many interface class examples as you can. We are low on real-world examples!

EdaphicStudio, 2018-10-15:
I have no experience with interface classes (though I like the concept!). So I'm coding blind here but found enough examples in the spec 8.26 to make a reasonable stab at rules.

It's **interesting** that the SystemVerilog Spec. Annex A only mentions the `interface_class_declaration` rule once! As a loose hanging rule! It's not part of the top `description` rule or any other rule (that I can find).

Here's a question: In practice can you legally declare an `interface class` anywhere you can declare a `class`? Or are there more limitations? I'm assuming it's legal anywhere a class can be declared for now.

edcote, 2018-10-15:
they are akin to Java interfaces

EdaphicStudio, 2018-10-15:
Yeah. It's a much-needed SV feature _but_ it also feels tacked on in the spec, and I haven't seen real-world examples of it. 

*Asking naively*: Do simulators support it now?

I'll do my best guess implementation based on your example and the examples in chapter 8.26 of the spec. 

Will start with just getting through the BNF (next release guaranteed) but will also need to work on resolve issues so we can check rules later.

edcote, 2018-10-15:
Yes.

https://static1.squarespace.com/static/521153a8e4b01a5565d75104/t/571d3b08746fb91673c11962/1461533459432/08_2.pdf
http://events.dvcon.org/2016/proceedings/papers/05_1.pdf
https://www.doulos.com/knowhow/sysverilog/tutorial/interface_classes/

EdaphicStudio, 2018-10-15:
Thank you!

I will make sure we have basic support out shortly (this week). That way we will at least not report errors! Then I will build in support for resolve etc after that.

EdaphicStudio, 2018-10-15:
**Would you be willing to try out an alpha version of a "rule support only" build _tomorrow_?**

I updated my rules to support `interface classes`, and it passes the examples in chapter 8.26 of the spec (see screenshot). Chapter 8.26 is now part of the regression (800+ tests). And the good news is that at least parsing these `interface classes` doesn't break anything else!

If you're willing to try an alpha, then I can upload a build here tomorrow that you can install locally. I will include instructions (it's easy!). The only *downside* is that you will have to manually delete this alpha build and reinstall from the plugin-stall once I release the final version. Let me know if you're interested.

Tomorrow and Wednesday I plan to add all examples from your links tomorrow, then focus on:

 1. resolve 
2. formatting
3. structure view support

Here is what my current local build looks like:

[image]

edcote, 2018-10-15:
sure,
e-mail directly at edmond.cote@gmail.com

EdaphicStudio, 2018-10-16:
I see two issues with your example above. You need to provide class name before parameter type declaration. Also, I don't think you can declare a local variable (`T m_strategy;`) in the class, at least I don't see it in the spec of the `interface_class_item`

I'm not saying this to nitpick. I haven't used interface classes and I'm worried I'm missing something!

[image]

EdaphicStudio, 2018-10-16:
@edcote I sent you an e-mail with instructions for how to test the alpha release for interface classes. It "should" work for correct code. It doesn't work as well as I want for error recovery and many other things. I'll keep working on more features in the background.

closed 2018-11-03 by EdaphicStudio