EdaphicStudio/SystemVerilog #23: comment line on `endif moved to next line during reformat

edcote, 2018-11-13:
Reformating the following:

```systemverilog
`ifndef HELLO
`define HELLO
`endif //!HELLO
```

Results in an inserted newline:

```systemverilog
`ifndef HELLO
`define HELLO
`endif
//!HELLO
```

EdaphicStudio, 2018-11-13:
> Reformating the following:
> `ifndef HELLO
> `define HELLO
> `endif //!HELLO
> Results in an inserted newline:
> `ifndef HELLO
> `define HELLO
> `endif
> //!HELLO

Nice find.

I just pushed a new BETA like 5 minutes ago that improves formatting. It seems that this update fixes this issue. It should be available within the next day or so.

In this next update I also format the expanded macros on hover:

Robert Ekendahl
Founder & Developer
Edaphic.Studio
p: +1 (617) 331-7554
e: robert@edaphic.studio

edcote, 2018-11-14:
Thank you! Closing. Will reopen if necessary.

closed 2018-11-14 by edcote