EdaphicStudio/SystemVerilog #25: Formatter inserts space when calling super.new

edcote, 2018-11-14:
This is on BETA 5.

```systemverilog
class c;
    function new (int a); // space after new
        super.new (a); // space after .new
    endfunction

    function void b(int c); // no space after b
        super.b(c); // no space after .b
    endfunction
endclass

/*
Expected:
class c;
    function new(int a);
        super.new(a);
    endfunction

    function void b(int c);
        super.b(c);
    endfunction
endclass
*/
```

EdaphicStudio, 2018-11-14:
This is an interesting issue. I was going back and forth on how to do this. One the one hand we should have spaces around keywords but on the other hand, we shouldn't have space before `(`. I settled on keyword overruling space-before-`(`. 

But I'm not sure which one is better. I'm happy to reverse it for the next release.

edcote, 2018-11-14:
*Imho*, space-after-`(` is typical for module declaration and instantiation. I tend to look at uvm library base for testbench code, i.e., no-space-after-`(`.

Not aware if there's an OO language that recommends no-space-after-`(` for function declarations and calls.

edcote, 2018-11-14:
* Proposal for design
```systemverilog
module a (
    input clock
);
    b b_inst (clock);
endmodule
```
* Proposal for testbench
```systemverilog
class c;
    function new();
        super.new()
    endfunction
endclass
````
... feel free to disregard, "arguing" about coding style is a zero-sum game ;)

EdaphicStudio, 2018-11-14:
Yeah, I'm not arguing as much as looking for some type of consistency that is close to what people expect. It's interesting but in the UVM UBUS examples they do what I'm doing currently (look as space around new):

https://github.com/rekendahl/uvm/blob/7d0d6c08e0e9325f808aa7c735e2ee0d650155c1/examples/integrated/ubus/examples/ubus_example_tb.sv

EdaphicStudio, 2018-11-14:
Long-term all these features will be options, but I'm looking to get down to some sane default formatting first. We're not there yet!

EdaphicStudio, 2019-09-15:
Fixed

closed 2019-09-15 by EdaphicStudio