EdaphicStudio/SystemVerilog #29: Clion reports: com.jetbrains.cidr.lang.psi.impl.OCFileImpl cannot be cast to studio.edaphic.systemverilog.psi.SVFile

apaiva, 2019-02-07:
CLion 2018.3.3
SystemVerilog v2019.1.BETA.3

CLion reports dozens of these firing in the background.  

```
java.lang.ClassCastException: com.jetbrains.cidr.lang.psi.impl.OCFileImpl cannot be cast to studio.edaphic.systemverilog.psi.SVFile
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.processIncludeFiles(NameResolution.kt:577)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.processLexicalDeclarations(NameResolution.kt:340)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.access$processLexicalDeclarations(NameResolution.kt:1)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt$processNestedScopesUpwards$2.invoke(NameResolution.kt:639)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt$processNestedScopesUpwards$2.invoke(NameResolution.kt)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.walkUp(NameResolution.kt:704)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.processNestedScopesUpwards(NameResolution.kt:610)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.processNestedScopesUpwards$default(NameResolution.kt:607)
	at studio.edaphic.systemverilog.resolve.NameResolutionKt.processReferenceIdentifierResolveVariants(NameResolution.kt:174)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceIdentifierReferenceImpl$resolveInner$1.invoke(SVReferenceIdentifierReferenceImpl.kt:19)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceIdentifierReferenceImpl$resolveInner$1.invoke(SVReferenceIdentifierReferenceImpl.kt:11)
	at studio.edaphic.systemverilog.resolve.ProcessorsKt.collectResolveVariants(Processors.kt:41)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceIdentifierReferenceImpl.resolveInner(SVReferenceIdentifierReferenceImpl.kt:19)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase$Resolver.resolve(SVReferenceBase.kt:71)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase$Resolver.resolve(SVReferenceBase.kt:69)
	at com.intellij.psi.impl.source.resolve.ResolveCache.lambda$resolve$0(ResolveCache.java:150)
	at com.intellij.openapi.util.RecursionManager$2.doPreventingRecursion(RecursionManager.java:98)
	at com.intellij.psi.impl.source.resolve.ResolveCache.resolve(ResolveCache.java:149)
	at com.intellij.psi.impl.source.resolve.ResolveCache.resolveWithCaching(ResolveCache.java:239)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase.advancedCachedMultiResolve(SVReferenceBase.kt:42)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase.multiResolve(SVReferenceBase.kt:35)
	at com.intellij.psi.PsiPolyVariantReferenceBase.resolve(PsiPolyVariantReferenceBase.java:47)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase.resolve(SVReferenceBase.kt:29)
	at studio.edaphic.systemverilog.resolve.ref.SVReferenceBase.resolve(SVReferenceBase.kt:22)
	at com.intellij.codeInsight.TargetElementUtil.doGetReferenceOrReferencedElement(TargetElementUtil.java:367)
	at com.intellij.codeInsight.TargetElementUtil.getReferenceOrReferencedElement(TargetElementUtil.java:345)
	at com.intellij.codeInsight.TargetElementUtil.doFindTargetElement(TargetElementUtil.java:219)
	at com.intellij.codeInsight.TargetElementUtil.findTargetElement(TargetElementUtil.java:191)
	at com.intellij.codeInsight.documentation.DocumentationManager.findTargetElementUnsafe(DocumentationManager.java:638)
	at com.intellij.codeInsight.documentation.DocumentationManager.findTargetElement(DocumentationManager.java:610)
	at com.intellij.codeInsight.documentation.QuickDocOnMouseOverManager$MyShowQuickDocRequest.lambda$run$0(QuickDocOnMouseOverManager.java:288)
	at com.intellij.openapi.application.impl.ApplicationImpl.tryRunReadAction(ApplicationImpl.java:1168)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.lambda$runInReadActionWithWriteActionPriority$0(ProgressIndicatorUtils.java:70)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.lambda$runWithWriteActionPriority$1(ProgressIndicatorUtils.java:123)
	at com.intellij.openapi.progress.impl.CoreProgressManager.lambda$runProcess$2(CoreProgressManager.java:164)
	at com.intellij.openapi.progress.impl.CoreProgressManager.registerIndicatorAndRun(CoreProgressManager.java:582)
	at com.intellij.openapi.progress.impl.CoreProgressManager.executeProcessUnderProgress(CoreProgressManager.java:532)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.executeProcessUnderProgress(ProgressManagerImpl.java:87)
	at com.intellij.openapi.progress.impl.CoreProgressManager.runProcess(CoreProgressManager.java:151)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.runWithWriteActionPriority(ProgressIndicatorUtils.java:112)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.runInReadActionWithWriteActionPriority(ProgressIndicatorUtils.java:70)
	at com.intellij.codeInsight.documentation.QuickDocUtil.runInReadActionWithWriteActionPriorityWithRetries(QuickDocUtil.java:91)
	at com.intellij.codeInsight.documentation.QuickDocOnMouseOverManager$MyShowQuickDocRequest.run(QuickDocOnMouseOverManager.java:286)
	at com.intellij.util.concurrency.QueueProcessor.runSafely(QueueProcessor.java:232)
	at com.intellij.util.Alarm$Request.runSafely(Alarm.java:356)
	at com.intellij.util.Alarm$Request.run(Alarm.java:343)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at com.intellij.util.concurrency.SchedulingWrapper$MyScheduledFutureTask.run(SchedulingWrapper.java:228)
	at com.intellij.util.concurrency.BoundedTaskExecutor.doRun(BoundedTaskExecutor.java:227)
	at com.intellij.util.concurrency.BoundedTaskExecutor.access$100(BoundedTaskExecutor.java:26)
	at com.intellij.util.concurrency.BoundedTaskExecutor$2$1.run(BoundedTaskExecutor.java:200)
	at com.intellij.util.ConcurrencyUtil.runUnderThreadName(ConcurrencyUtil.java:229)
	at com.intellij.util.concurrency.BoundedTaskExecutor$2.run(BoundedTaskExecutor.java:194)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```

pulzar, 2019-03-19:
I'm seeing the same problem when trying to 'jump to declaration' or Ctrl-click on any module. This is CLion on OSX.

EdaphicStudio, 2019-03-19:
Thanks for the update. This issue had fallen off the radar. Will try to address shortly. I have not tested directly against CLion but have been relying on JetBrains' automated checks. Clearly that is not good enough.

EdaphicStudio, 2019-09-15:
Believe this issue to be fixed

closed 2019-09-15 by EdaphicStudio